cirosantilli/linux-kernel-module-cheat #148: ModuleNotFoundError: No module named 'china_dictatorship'

MrZLeo, 2021-04-01:
When I run `./run`, I got :
```
Traceback (most recent call last):
  File "./run", line 12, in <module>
    import common
  File "/home/mrzleo/Desktop/linux-kernel-module-cheat/common.py", line 32, in <module>
    import china_dictatorship
ModuleNotFoundError: No module named 'china_dictatorship'
```

MrZLeo, 2021-04-01:
My device: MacBook Pro m1, using Ubuntu.20.04 in Parallels Desktop.

MrZLeo, 2021-04-01:
I find that `pip3 install china-dictatorship` works.

closed 2021-04-01 by MrZLeo

cirosantilli, 2021-04-01:
Yes, sorry about that, I updated that important module recently :-) In the docs e.g. https://cirosantilli.com/linux-kernel-module-cheat/#qemu-buildroot-setup-getting-started this is getting installed at `./setup` BTW, let me know if I missed mentioning it in any getting started setup.