cirosantilli/linux-kernel-module-cheat #149: E: Unable to locate package rr

MrZLeo, 2021-04-01:
when I use: `./build --download-dependencies qemu-buildroot`, I got:
```
E: Unable to locate package rr
Traceback (most recent call last):
  File "./build", line 680, in <module>
    Main().cli()
  File "/home/mrzleo/Desktop/linux-kernel-module-cheat/cli_function.py", line 267, in cli
    exit_status = self.cli_noexit(*args, **kwargs)
  File "/home/mrzleo/Desktop/linux-kernel-module-cheat/cli_function.py", line 258, in cli_noexit
    return self._do_main(vars(args))
  File "/home/mrzleo/Desktop/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/mrzleo/Desktop/linux-kernel-module-cheat/common.py", line 1540, in main
    ret = self.timed_main()
  File "./build", line 566, in timed_main
    self.sh.run_cmd(
  File "/home/mrzleo/Desktop/linux-kernel-module-cheat/shell_helpers.py", line 457, in run_cmd
    raise e
Exception: Command exited with status: 100
```

cirosantilli, 2021-04-01:
What's your OS and version? Remove rr from `build` as a workaround.

MrZLeo, 2021-04-02:
> What's your OS and version? Remove rr from `build` as a workaround, or use the Docker Ubuntu 20.04 build: https://cirosantilli.com/linux-kernel-module-cheat/#docker

I am in Ubuntu 20.04, remove rr in `build` still can't work properly, error here is:
```
The following packages have unmet dependencies:
 builddeps:qemu : Depends: gcc-alpha-linux-gnu but it is not installable
                  Depends: gcc-powerpc64-linux-gnu but it is not installable
E: Unable to correct problems, you have held broken packages.
Traceback (most recent call last):
  File "./build", line 679, in <module>
    Main().cli()
  File "/home/mrzleo/Desktop/linux-kernel-module-cheat/cli_function.py", line 267, in cli
    exit_status = self.cli_noexit(*args, **kwargs)
  File "/home/mrzleo/Desktop/linux-kernel-module-cheat/cli_function.py", line 258, in cli_noexit
    return self._do_main(vars(args))
  File "/home/mrzleo/Desktop/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/mrzleo/Desktop/linux-kernel-module-cheat/common.py", line 1540, in main
    ret = self.timed_main()
  File "./build", line 570, in timed_main
    self.sh.run_cmd(
  File "/home/mrzleo/Desktop/linux-kernel-module-cheat/shell_helpers.py", line 457, in run_cmd
    raise e
Exception: Command exited with status: 100

```
It seems a bug of ubuntu 20.04, you can see here: https://bugs.launchpad.net/ubuntu/+source/qemu/+bug/1866032
dose it mean that I can not use 20.04?

cirosantilli, 2021-04-02:
The reference build is 20.04 in Docker: https://cirosantilli.com/linux-kernel-module-cheat/#docker so there must be something that is different between your system and Docker.

That issue appears to happen only on non x86 host ISAs, what's your host ISA?

That would also explain why you have no `rr`. I haven't tested much on non-x86 hosts, and wouldn't be surprised if something is not working there, though in theory it should not be hard to port. Have you tried the workaround proposed in that issue, modify commands in `./build` to do a:

```
apt build-dep --arch-only qemu
```

MrZLeo, 2021-04-02:
> The reference build is 20.04 in Docker: https://cirosantilli.com/linux-kernel-module-cheat/#docker so there must be something that is different between your system and Docker.
>
> That issue appears to [...]

I am in m1 which based on aarch64, and `apt build-dep --arch-only qemu` can work but after that `./run` will raise an error.

I think that must be the problem of architecture.

cirosantilli, 2021-04-02:
> I am in m1 which based on aarch64, and `apt build-dep --arch-only qemu` can work but after that `./run` will raise an error.

What's the next error message? There's no fundamental reason why it won't work, if you keep giving me error messages I will keep trying to reply, it would good to have it working on other ISAs :-)

MrZLeo, 2021-04-02:
> What's the next error message? There's no fundamental reason why it won't work, if you keep giving me error messages I will keep trying to reply, it would good to have it working on other ISAs :-)

if `./run`, then will get this:
```
Traceback (most recent call last):
  File "./run", line 999, in <module>
    Main().cli()
  File "/home/mrzleo/Desktop/linux-kernel-module-cheat/cli_function.py", line 267, in cli
    exit_status = self.cli_noexit(*args, **kwargs)
  File "/home/mrzleo/Desktop/linux-kernel-module-cheat/cli_function.py", line 258, in cli_noexit
    return self._do_main(vars(args))
  File "/home/mrzleo/Desktop/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/mrzleo/Desktop/linux-kernel-module-cheat/common.py", line 1540, in main
    ret = self.timed_main()
  File "./run", line 483, in timed_main
    raise_image_not_found()
  File "./run", line 459, in raise_image_not_found
    raise Exception('Executable image not found. Did you build it? ' \
Exception: Executable image not found. Did you build it? Tried to use: /home/mrzleo/Desktop/linux-kernel-module-cheat/out/linux/default/x86_64/arch/x86/boot/bzImage
```

thanks a lot!

MrZLeo, 2021-04-02:
> if `./run`, then will get this:
>
> ```
> Traceback (most recent call last):
>   File "./run", line 999, in <module>
>     Main().cli()
>   File [...]

It seems build don't work, but `sudo apt build-dep --arch-only qemu` don't show any error:
```
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
```

cirosantilli, 2021-04-02:
> It seems build don't work, but sudo apt build-dep --arch-only qemu don't show any error:

Yes, the image didn't get built. The logs you show have no errors. There should be an error below I'd expect as the rest of the build gets done.

MrZLeo, 2021-04-02:
> > It seems build don't work, but sudo apt build-dep --arch-only qemu don't show any error:
> 
> Yes, the image didn't get built. The logs you show have no errors. There should be an error below I'd expect as the rest of the build gets done.

but no, the only logs is that by using `sudo apt build-dep --arch-only qemu`...

mjkpolo, 2021-05-22:
I have the same error but I'm a total noob who has never done any kernel dev just trying to learn. Also on m1 and I ran `sudo apt build-dep --arch-only qemu`

vignesh0025, 2023-10-23:
Can verify this error exist in docker env in MAC M2 with Ubuntu 20.04 as well. Please share a solution for the same

cirosantilli, 2023-10-23:
Easy fix is:

```
diff --git a/Dockerfile b/Dockerfile
index 94766f1..9ae8dca 100644
--- a/Dockerfile
+++ b/Dockerfile
@@ -1,5 +1,6 @@
 # https://cirosantilli.com/linux-kernel-module-cheat#docker
 FROM ubuntu:20.04
+RUN touch /.dockerenv
 COPY setup /
 COPY requirements.txt /
 RUN /setup -y
```

We need to add a host arch check.

closed 2023-10-23 by cirosantilli

vignesh0025, 2023-10-24:
That fixed the rr package issue but unfortunately, the other two issues mentioned above exists still.

"The following packages have unmet dependencies:"

and 

"Exception: Executable image not found. Did you build it? Tried to use:"

cirosantilli, 2023-10-26:
> That fixed the rr package issue but unfortunately, the other two issues mentioned above exists still.
> 
> "The following packages have unmet dependencies:"
> 

This is a harder issue. If you don't compile QEMU from source, there's a chance you'll use an incompatible version. 

There is a `./run --qemu-which host` if you want to try with your OS provided QEMU anyways. 

You also need to remove `'qemu` from:

```
diff --git a/build b/build
index ae3fbfb..694228a 100755
--- a/build
+++ b/build
@@ -351,7 +351,6 @@ so looping over all of them would waste time.
             ),
             'qemu': _Component(
                 self._build_file('build-qemu'),
-                apt_build_deps={'qemu'},
                 apt_get_pkgs={
                     'libsdl2-dev',
                     'ninja-build',
```

ideally the `--qemu-which host`  flag should do that but it doesn't currently.

> and
> 
> "Exception: Executable image not found. Did you build it? Tried to use:"

This is just because the build failed previously.