cirosantilli/linux-kernel-module-cheat #150: libstdc++.la fails with multiple definition error

MathewRo, 2021-05-08:
Hi Ciro,

./build --download-dependencies qemu-buildroot fails with the below error for me
I'm currently working on WSL2.
There were other errors related to case sensitive filesystem that I could resolve. This one however, I'm not sure how to solve it as it seems to arise from linking  libstdc++.la

'/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/x86_64-buildroot-linux-gnu/libstdc++-v3/src'
/bin/bash ../libtool --tag CXX   --mode=link /mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/./gcc/xgcc -shared-libgcc -B/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/./gcc -nostdinc++ -L/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/x86_64-buildroot-linux-gnu/libstdc++-v3/src -L/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/x86_64-buildroot-linux-gnu/libstdc++-v3/src/.libs -L/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/x86_64-buildroot-linux-gnu/libstdc++-v3/libsupc++/.libs -B/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/bin/ -B/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/lib/ -isystem /mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/include -isystem /mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/sys-include     -Wl,-O1 -Wl,-z,relro -Wl,--gc-sections  -std=gnu++98 -fPIC -DPIC -fno-implicit-templates  -Wall -Wextra -Wwrite-strings -Wcast-qual -Wabi  -fdiagnostics-show-location=once   -ffunction-sections -fdata-sections  -frandom-seed=libstdc++.la  -o libstdc++.la -version-info 6:25:0 -Wl,--version-script=libstdc++-symbols.ver -lm -rpath /mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/lib/../lib64 compatibility.lo compatibility-debug_list.lo compatibility-debug_list-2.lo  compatibility-c++0x.lo compatibility-atomic-c++0x.lo compatibility-thread-c++0x.lo compatibility-chrono.lo compatibility-condvar.lo  ../libsupc++/libsupc++convenience.la ../src/c++98/libc++98convenience.la ../src/c++11/libc++11convenience.la
libtool: link:  /mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/./gcc/xgcc -shared-libgcc -B/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/./gcc -nostdinc++ -L/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/x86_64-buildroot-linux-gnu/libstdc++-v3/src -L/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/x86_64-buildroot-linux-gnu/libstdc++-v3/src/.libs -L/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/x86_64-buildroot-linux-gnu/libstdc++-v3/libsupc++/.libs -B/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/bin/ -B/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/lib/ -isystem /mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/include -isystem /mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/sys-include     -fPIC -DPIC -D_GLIBCXX_SHARED -shared -nostdlib /mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/../lib64/crti.o /mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/./gcc/crtbeginS.o  .libs/compatibility.o .libs/compatibility-debug_list.o .libs/compatibility-debug_list-2.o .libs/compatibility-c++0x.o .libs/compatibility-atomic-c++0x.o .libs/compatibility-thread-c++0x.o .libs/compatibility-chrono.o .libs/compatibility-condvar.o  -Wl,--whole-archive ../libsupc++/.libs/libsupc++convenience.a ../src/c++98/.libs/libc++98convenience.a ../src/c++11/.libs/libc++11convenience.a -Wl,--no-whole-archive  -L/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/x86_64-buildroot-linux-gnu/libstdc++-v3/libsupc++/.libs -L/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/x86_64-buildroot-linux-gnu/libstdc++-v3/src -L/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/x86_64-buildroot-linux-gnu/libstdc++-v3/src/.libs -lm -L/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/./gcc -L/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/bin -L/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/lib -L/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/sysroot/lib/../lib64 -L/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/../lib64 -L/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/sysroot/lib -L/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib -lc -lgcc_s /mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/./gcc/crtendS.o /mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib/../lib64/crtn.o  -Wl,-O1 -Wl,-z -Wl,relro -Wl,--gc-sections -Wl,--version-script=libstdc++-symbols.ver   -Wl,-soname -Wl,libstdc++.so.6 -o .libs/libstdc++.so.6.0.25
/usr/bin/make  all-am
make[4]: Entering directory '/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/x86_64-buildroot-linux-gnu/libgfortran'
true  DO=all multi-do # /usr/bin/make
:
:
:
make[4]: Leaving directory '/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/x86_64-buildroot-linux-gnu/libgfortran'
make[3]: Leaving directory '/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/x86_64-buildroot-linux-gnu/libgfortran'
/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/bin/ld: /mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/sysroot/usr/lib64/libc_nonshared.a(init-first.os): in function `_init':
[... cut ...]
make[1]: Leaving directory '/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build'
package/pkg-generic.mk:247: recipe for target '/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/.stamp_built' failed
make: *** [/mnt/c/Users/mathewro/ubuntu/research/github/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-gcc-final-custom/.stamp_built] Error 2

cirosantilli, 2021-05-08:
Rohit, I'm afraid I don't know exactly what is going on.

The best I can recommend is the usual generic approach of "minimize the problem to a minimal Buildroot-only issue" (since this happens while building the cross compiler, e.g. by copying out the config and redoing a build on a clean Buildroot tree), and then attack that in isolation.

Have a look to see if any commits in newer Buildroot seem to fix it (happens often), and if not send them a bug request (and patches for the other stuff you've fixed! :-))

Also consider using the Ubuntu 20.04 Docker/VM setup if the above does not yield results.

MathewRo, 2021-05-08:
Thanks Ciro for the quick update. Right now I'm bringing up the docker setup :) I'll keep you posted if I can get the native one to work!