cirosantilli/linux-kernel-module-cheat #152: Related to Qemu Emulation.

Romit188, 2021-05-21:
I am using the link, which is provided by you for build and boot kernel on qemu.

From this I am using the kernel configurations for arm architecture.
Which is shown in below.
64-bit kernel ---> yes
General setup ---> Initial RAM filesystem and RAM disk (initramfs/initrd) support ---> yes
General setup ---> Configure standard kernel features ---> Enable support for printk ---> yes
Executable file formats / Emulations ---> Kernel support for ELF binaries ---> yes
Executable file formats / Emulations ---> Kernel support for scripts starting with #! ---> yes
Device Drivers ---> Generic Driver Options ---> Maintain a devtmpfs filesystem to mount at /dev ---> yes
Device Drivers ---> Generic Driver Options ---> Automount devtmpfs at /dev, after the kernel mounted the rootfs ---> yes
Device Drivers ---> Character devices ---> Enable TTY ---> yes
Device Drivers ---> Character devices ---> Serial drivers ---> 8250/16550 and compatible serial support ---> yes
Device Drivers ---> Character devices ---> Serial drivers ---> Console on 8250/16550 and compatible serial port ---> yes
File systems ---> Pseudo filesystems ---> /proc file system support ---> yes
File systems ---> Pseudo filesystems ---> sysfs file system support ---> yes

And the compile the kernel using below command.
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabi- install

And i get zImage successfully.

Then I try to boot linux kernel for V-express board on qemu using below command.

qemu-system-arm -M vexpress-a9 -dtb ./arch/arm/boot/dts/vexpress-v2p-ca9.dtb -kernel ./arch/arm/boot/zImage -append "root=/dev/mmcblk0 console=ttyAMA0" -sd /home/romit/SourceCode/busybox-1.33.1/RootFS.ext3 -serial stdio 

I get error , Which is shown in below.

WARNING: Image format was not specified for '/home/romit/SourceCode/busybox-1.33.1/RootFS.ext3' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.
pulseaudio: set_sink_input_volume() failed
pulseaudio: Reason: Invalid argument
pulseaudio: set_sink_input_mute() failed
pulseaudio: Reason: Invalid argument

I don't understand this problem, Please any one can help about the same.

cirosantilli, 2021-05-23:
Romit, I don't see anything on the error message that is likely to be the cause of the problem. Can you provide fuller logs if there's anything else? So QEMU dies and you get no shell?

Romit188, 2021-05-25:
Hello cirosantilli,

Thank you for valuable reply...
I successfully boot the kernel on qemu.
So, I close the issue right now.

closed 2021-05-25 by Romit188