cirosantilli/linux-kernel-module-cheat #156: submodule qemu/tests/fp/meson.build:27:0: ERROR: Include dir berkeley-softfloat-3/source/include does not exist.

rahulravi26081990, 2021-07-25:
Hi,
I am trying to build using below command.
./build --arch aarch64 --download-dependencies qemu-buildroot

**But getting the below error.**
**../../../../submodules/qemu/tests/fp/meson.build:27:0: ERROR: Include dir berkeley-softfloat-3/source/include does not exist.**

A full log can be found at /home/rahulravi26/qemu-buildroot/linux-kernel-module-cheat/out/qemu/default/opt/meson-logs/meson-log.txt

ERROR: meson setup failed

Traceback (most recent call last):
  File "./build", line 680, in <module>
    Main().cli()
  File "/home/rahulravi26/qemu-buildroot/linux-kernel-module-cheat/cli_function.py", line 267, in cli
    exit_status = self.cli_noexit(*args, **kwargs)
  File "/home/rahulravi26/qemu-buildroot/linux-kernel-module-cheat/cli_function.py", line 258, in cli_noexit
    return self._do_main(vars(args))
  File "/home/rahulravi26/qemu-buildroot/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/rahulravi26/qemu-buildroot/linux-kernel-module-cheat/common.py", line 1540, in main
    ret = self.timed_main()
  File "./build", line 675, in timed_main
    ret = component.build(self.env['arch'])
  File "./build", line 54, in build
    return self.build_callback()
  File "./build", line 473, in f
    return lkmc.import_path.import_path_main(component_file)(**args)
  File "/home/rahulravi26/qemu-buildroot/linux-kernel-module-cheat/common.py", line 737, in __call__
    return super().__call__(**kwargs)
  File "/home/rahulravi26/qemu-buildroot/linux-kernel-module-cheat/cli_function.py", line 149, in __call__
    return self._do_main(kwargs)
  File "/home/rahulravi26/qemu-buildroot/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/rahulravi26/qemu-buildroot/linux-kernel-module-cheat/common.py", line 1540, in main
    ret = self.timed_main()
  File "/home/rahulravi26/qemu-buildroot/linux-kernel-module-cheat/common.py", line 2059, in timed_main
    return self.build()
  File "/home/rahulravi26/qemu-buildroot/linux-kernel-module-cheat/build-qemu", line 38, in build
    self.sh.run_cmd(
  File "/home/rahulravi26/qemu-buildroot/linux-kernel-module-cheat/shell_helpers.py", line 457, in run_cmd
    raise e
Exception: Command exited with status: 1

cirosantilli, 2021-07-25:
Hmm, not too sure what is going on.

On my machine `./submodules/qemu/tests/fp/berkeley-softfloat-3/` exists, does it not exist for you? 

That directory a submodule of QEMU as seen on `.gitmodules`:

```
[submodule "tests/fp/berkeley-testfloat-3"]
	path = tests/fp/berkeley-testfloat-3
	url = https://git.qemu.org/git/berkeley-testfloat-3.git
```

so I would try:

```
cd submodules/qemu
git submodule init --update
```

I'm going to try a Docker build as well.

rahulravi26081990, 2021-07-25:
./submodules/qemu/tests/fp/berkeley-softfloat-3/ exits on my workspace too.

But inside that only .git file exits having content of **gitdir: ../../../../../.git/modules/submodules/qemu/modules/tests/fp/berkeley-softfloat-3**

rahulravi26@ubuntu2004-VirtualBox:~/qemu-buildroot/linux-kernel-module-cheat$ ls -la submodules/qemu/tests/fp/berkeley-softfloat-3/
total 12
drwxrwxr-x 2 rahulravi26 rahulravi26 4096 Jul 25 13:41 .
drwxrwxr-x 4 rahulravi26 rahulravi26 4096 Jul 25 08:49 ..
-rw-rw-r-- 1 rahulravi26 rahulravi26   90 Jul 24 14:24 .git

rahulravi26@ubuntu2004-VirtualBox:~/qemu-buildroot/linux-kernel-module-cheat$ ls -l .git/modules/submodules/qemu/modules/tests/fp/berkeley-softfloat-3/
total 40
drwxrwxr-x 2 rahulravi26 rahulravi26 4096 Jul 24 14:24 branches
-rw-rw-r-- 1 rahulravi26 rahulravi26  356 Jul 25 13:37 config
-rw-rw-r-- 1 rahulravi26 rahulravi26   73 Jul 24 14:24 description
-rw-rw-r-- 1 rahulravi26 rahulravi26   23 Jul 24 14:24 HEAD
drwxrwxr-x 2 rahulravi26 rahulravi26 4096 Jul 24 14:24 hooks
drwxrwxr-x 2 rahulravi26 rahulravi26 4096 Jul 24 14:24 info
drwxrwxr-x 3 rahulravi26 rahulravi26 4096 Jul 24 14:24 logs
drwxrwxr-x 4 rahulravi26 rahulravi26 4096 Jul 24 14:24 objects
-rw-rw-r-- 1 rahulravi26 rahulravi26  114 Jul 24 14:24 packed-refs
drwxrwxr-x 5 rahulravi26 rahulravi26 4096 Jul 24 14:24 refs

cirosantilli, 2021-07-25:
OK. I did the Docker build right now https://github.com/cirosantilli/linux-kernel-module-cheat/tree/38eb67ad2ec46b58dbc701073ce0e980fad7788b#docker with `--arch aarch64` on a fully clean repo and it just worked, so it isn't something I can reproduce, I can only give the unsatisfactory guess that the submodule update failed for a transient network reason.

Try to work around by using `git submodule init --update` manually inside `submodules/qemu` to ensure the submodule is update, and then try running again. Let me know if you find anything else.

rahulravi26081990, 2021-07-27:
Fresh clean build actually solve the issue. Thanks for your support.

Do you have any command to clean the complete repo, instead of making fresh build?

cirosantilli, 2021-07-27:
> Fresh clean build actually solve the issue. Thanks for your support.
> 
> Do you have any command to clean the complete repo, instead of making fresh build?

The build is under `out/`, so `rm -rf out` cleans it. But in your case it was a submodule issue, so not sure what's the cleanest approach, I think this works:

```
rm -rf submodules
git checkout -- submodules
```

closed 2021-07-30 by rahulravi26081990