cirosantilli/linux-kernel-module-cheat #159: LLVM cross-compile userland?

4fterlook, 2021-08-28:
Hello,

I was trying to compile my own benchmark with gem5-buildroot aarch64 full system mode. This is what I did.

`./build --arch aarch64 gem5-buildroot`
`./run --arch aarch64 --emulator gem5 --eval-after 'm5 checkpoint;'`

And then I connect with gem5-shell

I found this would automatically compile every file under userland/ and it's really great. However, I cannot find the compile scripts for these userland files. I want to change the compiler to llvm because I'm trying to do experiments related to Polly project of LLVM. Could this be done?

cirosantilli, 2021-08-28:
Hi, the key userland build entry point is: https://github.com/cirosantilli/linux-kernel-module-cheat/blob/1a1237218e6cef80419b051c763ef4785638669e/build-userland but like most scripts it calls stuff from https://github.com/cirosantilli/linux-kernel-module-cheat/blob/1a1237218e6cef80419b051c763ef4785638669e/common.py

If you hack `gcc_path` to point to clang, it will likely work. A clean patch would likely be adding a new `--gcc-which=llvm`.

4fterlook, 2021-08-28:
Thank you so much! I'm trying to use clang to cross-compile instead, yet I'm not familiar with buildroot. It seems that out/buildroot/build only has the headers for c++. May I ask where could I find the headers and .so for aarch64? Like stdio.h and libgomp.so?

cirosantilli, 2021-08-28:
Buildroot doesn't support clang, so the approach I propose is to just use your host distro (Ubuntu) provided aarch64 clang. `--gcc-which host` already does that for GCC, the quick hack path is likely to add that option, and hack the `gcc_path` variable I mentioned in the previous comment.

But the GCC aarch64 headers and stdlib .so should be to be under the buildroot build. But I don't have any builds now to find the path. Something like `cd out/buildroot/build/default/aarch64` and `find . -iname ...` should give them.

4fterlook, 2021-08-28:
Thank you, I'll try using the toolchain provided by [Linaro](https://www.linaro.org/downloads/). I'll create a patch if that works.

closed 2021-08-28 by 4fterlook