cirosantilli/linux-kernel-module-cheat #160: Some errors when build on ubuntu 20.04 (installed via virtualbox on windows)

jundai20, 2021-10-18:
Tested on a brand new ubuntu 20.04 on Virutalbox, meet following issues:

./build --download-dependencies qemu-buildroot

1.  Need to modify  /etc/apt/sources.list to add back deb-src
==> uncomment deb-src

2. Need to manually install following packages
 builddeps:qemu : Depends: libgtk-3-dev but it is not going to be installed
                  Depends: libvte-2.91-dev but it is not going to be installed
                  Depends: libepoxy-dev but it is not going to be installed
                  Depends: libgbm-dev but it is not going to be installed

3. The openssl git name has a typo. Instead of openssl, it should be openssl.git
qemu/roms/edk2/.gitmodules ==> openssl-> openssl.git

4. It will kill ubuntu GUI mode, smash the host linux
After ./build --arch aarch64 --download-dependencies qemu-buildroot, the ubuntu GUI is disspear. Still can login, but it is CLI mode.

5. Pending, DO not know what happened, give up here :)
Later, the host  out/buildroot/buid/default/host-uitl-linux-2.35.2  will fail, the error is, 
configure: error: C compiler cannot create executables

cirosantilli, 2021-10-18:
1 and 2: discussed at https://cirosantilli.com/linux-kernel-module-cheat/put-source-uris-in-sources which is automated in Docker: https://github.com/cirosantilli/linux-kernel-module-cheat/blob/0acd34a1d4bceb94de634390af605e6d67e0abbf/build#L552

3. I suspect that both openssl and openssl.git work since they are GitHub URLs and you can `git clone` either, and that instead it was a one off internet/github problem. I have also been able to checkout that version of QEMU just now.

4. 5. Not sure about those.

jundai20, 2021-10-19:
Thanks for the information.  Yes, I created a 42G VirtualBox fixed size vdi for this purpose.

BTW, follow the following link, I've brought back the GUI. There should be some gdm version confliction.
https://askubuntu.com/questions/641642/gui-does-not-start
sudo apt purge gdm gdm3
sudo apt install gdm3 ubuntu-desktop
systemctl restart gdm