cirosantilli/linux-kernel-module-cheat #170: 新疆维吾尔再教育營迫害 视频照片 Xinjiang Uyghur re-education centers persecution videos and photos

cirosantilli, 2022-04-01:
# 新疆维吾尔族人在公共汽车上被迫流离失所 Forced displacement of Uyghurs from Xinjiang across China in busses (2020)

https://www.codastory.com/authoritarian-tech/china-uyghur-migration/

https://cirosantilli.com/china-dictatorship/#forced-displacement-of-uyghurs-across-china-2020

https://user-images.githubusercontent.com/1429315/114305864-d78b3900-9ad1-11eb-9b23-be4db4ae1fa7.mp4

https://user-images.githubusercontent.com/1429315/114305839-c3473c00-9ad1-11eb-994a-e553d80181e9.mp4

https://user-images.githubusercontent.com/1429315/114305828-b6c2e380-9ad1-11eb-96f0-b65910026d24.mp4

https://user-images.githubusercontent.com/1429315/114305847-ce9a6780-9ad1-11eb-80cd-0ed9ffc1cbc6.mp4

# Merdan Ghappar's prison video (2020, 麥爾丹·阿巴的监狱视频)

https://cirosantilli.com/china-dictatorship/#merdan-ghappars-video-2020

https://user-images.githubusercontent.com/1429315/122738464-e92f5080-d279-11eb-8b96-5524d179b11a.mp4

https://user-images.githubusercontent.com/1429315/122738573-0532f200-d27a-11eb-8494-0f9aaf78cf0f.mp4

# 社会主义核心价值观，一定维护民族团结 Uyghur brainwashing: Socialist core values maintain ethnic harmony UNCONFIRMED

https://user-images.githubusercontent.com/1429315/114300978-920f4180-9aba-11eb-903d-23a145b1c5ba.mp4

https://www.youtube.com/watch?v=AxnS4z-cSlI

https://twitter.com/__Inty__/status/1381002964551168000

https://cirosantilli.com/china-dictatorship/#xinjiang-re-education-camps

#  维吾尔人在拘留营唱着快乐的歌 Uyghurs singing a happy song at internment camp 

https://www.youtube.com/watch?t=432&v=WmId2ZP3h0c

https://cirosantilli.com/china-dictatorship/#xinjiang-re-education-camps

https://user-images.githubusercontent.com/1429315/103039458-deb12a00-4568-11eb-9281-7cb8550666ca.mp4

#  2018年维吾尔族人在喀什营地打乒乓球跳快乐舞 Uyghurs playing ping pong and dancing a happy dance in a Kashgar camp in 2018

https://mobile.twitter.com/HuXijin_GT/status/1055013584806256644  by Hu Xijin 胡锡进

https://user-images.githubusercontent.com/1429315/130763853-b4541f8f-1b62-46ba-b617-f4ca60fbeae7.mp4

https://user-images.githubusercontent.com/1429315/130763814-33f8b836-ae8a-4764-892a-496430844378.mp4

# Uyghur children marched with large bags UNCONFIRMED, @MaD DaN, 伯干村 Bogancun, Xinjiang 诗书达理之山人刨作原声

https://user-images.githubusercontent.com/1429315/114303176-b6701b80-9ac4-11eb-8c27-38e2413209ab.mp4

https://twitter.com/Uyghurian/status/1380992770823835664

https://cirosantilli.com/china-dictatorship/#xinjiang-re-education-camps

# Teacher in Xinjiang slaps Uyghur student on the face 11 times 新疆老师打维族学生11个耳光 （建设美丽新疆）(UNCONFIRMED)

https://user-images.githubusercontent.com/1429315/117796953-7331d580-b247-11eb-9af8-6042470db542.mp4

https://www.reddit.com/r/China/comments/n99fe9/teacher_in_xinjiang_school_slaps_a_uyghur_student/

This video cannot be uploaded to YouTube because of their child safety community guidelines. It was automatically flagged, and my appeal was declined by a human. Basically it's bad for advertisers, and some politician might get sad. They prefer dictatorship to truth.

# 新疆维吾尔再教育營 Xinjiang re-education camps

https://cirosantilli.com/china-dictatorship/#xinjiang

https://cirosantilli.com/china-dictatorship/#xinjiang-re-education-camps

Xinjiang Data Project screenshot of camp map 2021 jpg

[image]

Xinjiang camp in camp jpg

[image]

Uyghur playing guitar jpg

[image]

Xinjiang prisoners sitting identified jpeg

[image]

Dabancheng reeducation camp jpg

[image]

Artux city reeducation camp webp

[image]

Uyghur sale website jpg

[image]

Uyghur workers waving China flag jpg

[image]

Xinjiang cotton vs American plantation from 1908 jpg

[image]

Racism in the West vs China George Floyd vs Xinjiang jpg

[image]

Xinjiang I won jpeg

[image]

https://cirosantilli.com/china-dictatorship/#holocaust-denial

Blood Cotton Initiative by Wuheqilin holocaust denial jpg

[image]

# Xinjiang outside reeducation camps

Xinjiang chained butcher knives jpg

[image]

AIGZ123, 2022-04-09:
你们这群反华的狗垃圾是真的恶心该死！

lanekkk233, 2022-04-11:
看不懂中文还瞎几把发 搞笑

cirosantilli, 2022-04-11:
test

cirosantilli, 2022-04-11:
# 新疆维吾尔族人在公共汽车上被迫流离失所 Forced displacement of Uyghurs from Xinjiang across China in busses (2020)

https://www.codastory.com/authoritarian-tech/china-uyghur-migration/

https://cirosantilli.com/china-dictatorship/#forced-displacement-of-uyghurs-across-china-2020

https://user-images.githubusercontent.com/1429315/114305864-d78b3900-9ad1-11eb-9b23-be4db4ae1fa7.mp4

https://user-images.githubusercontent.com/1429315/114305839-c3473c00-9ad1-11eb-994a-e553d80181e9.mp4

https://user-images.githubusercontent.com/1429315/114305828-b6c2e380-9ad1-11eb-96f0-b65910026d24.mp4

https://user-images.githubusercontent.com/1429315/114305847-ce9a6780-9ad1-11eb-80cd-0ed9ffc1cbc6.mp4

# Merdan Ghappar's prison video (2020, 麥爾丹·阿巴的监狱视频)

https://cirosantilli.com/china-dictatorship/#merdan-ghappars-video-2020

https://user-images.githubusercontent.com/1429315/122738464-e92f5080-d279-11eb-8b96-5524d179b11a.mp4

https://user-images.githubusercontent.com/1429315/122738573-0532f200-d27a-11eb-8494-0f9aaf78cf0f.mp4

# 社会主义核心价值观，一定维护民族团结 Uyghur brainwashing: Socialist core values maintain ethnic harmony UNCONFIRMED

https://user-images.githubusercontent.com/1429315/114300978-920f4180-9aba-11eb-903d-23a145b1c5ba.mp4

https://www.youtube.com/watch?v=AxnS4z-cSlI

https://twitter.com/__Inty__/status/1381002964551168000

https://cirosantilli.com/china-dictatorship/#xinjiang-re-education-camps

#  维吾尔人在拘留营唱着快乐的歌 Uyghurs singing a happy song at internment camp 

https://www.youtube.com/watch?t=432&v=WmId2ZP3h0c

https://cirosantilli.com/china-dictatorship/#xinjiang-re-education-camps

https://user-images.githubusercontent.com/1429315/103039458-deb12a00-4568-11eb-9281-7cb8550666ca.mp4

#  2018年维吾尔族人在喀什营地打乒乓球跳快乐舞 Uyghurs playing ping pong and dancing a happy dance in a Kashgar camp in 2018

https://mobile.twitter.com/HuXijin_GT/status/1055013584806256644  by Hu Xijin 胡锡进

https://user-images.githubusercontent.com/1429315/130763853-b4541f8f-1b62-46ba-b617-f4ca60fbeae7.mp4

https://user-images.githubusercontent.com/1429315/130763814-33f8b836-ae8a-4764-892a-496430844378.mp4

# Uyghur children marched with large bags UNCONFIRMED, @MaD DaN, 伯干村 Bogancun, Xinjiang 诗书达理之山人刨作原声

https://user-images.githubusercontent.com/1429315/114303176-b6701b80-9ac4-11eb-8c27-38e2413209ab.mp4

https://twitter.com/Uyghurian/status/1380992770823835664

https://cirosantilli.com/china-dictatorship/#xinjiang-re-education-camps

# Teacher in Xinjiang slaps Uyghur student on the face 11 times 新疆老师打维族学生11个耳光 （建设美丽新疆）(UNCONFIRMED)

https://user-images.githubusercontent.com/1429315/117796953-7331d580-b247-11eb-9af8-6042470db542.mp4

https://www.reddit.com/r/China/comments/n99fe9/teacher_in_xinjiang_school_slaps_a_uyghur_student/

This video cannot be uploaded to YouTube because of their child safety community guidelines. It was automatically flagged, and my appeal was declined by a human. Basically it's bad for advertisers, and some politician might get sad. They prefer dictatorship to truth.

# 新疆维吾尔再教育營 Xinjiang re-education camps

https://cirosantilli.com/china-dictatorship/#xinjiang

https://cirosantilli.com/china-dictatorship/#xinjiang-re-education-camps

Xinjiang Data Project screenshot of camp map 2021 jpg

[image]

Xinjiang camp in camp jpg

[image]

Uyghur playing guitar jpg

[image]

Xinjiang prisoners sitting identified jpeg

[image]

Dabancheng reeducation camp jpg

[image]

Artux city reeducation camp webp

[image]

Uyghur sale website jpg

[image]

Uyghur workers waving China flag jpg

[image]

Xinjiang cotton vs American plantation from 1908 jpg

[image]

Racism in the West vs China George Floyd vs Xinjiang jpg

[image]

Xinjiang I won jpeg

[image]

https://cirosantilli.com/china-dictatorship/#holocaust-denial

Blood Cotton Initiative by Wuheqilin holocaust denial jpg

[image]

# Xinjiang outside reeducation camps

Xinjiang chained butcher knives jpg

[image]

9PEAK, 2022-05-10:
I have a question that, how to post these sensitive news?
R u in China.
In my opinion, if u r in China, u should git the info to your own server outside Mainland, then git the info from outside to github. All git transfer must be https.
Am I right?

cirosantilli, 2022-05-10:
> I have a question that, how to post these sensitive news? R u in China. In my opinion, if u r in China, u should git the info to your own server outside Mainland, then git the info from outside to github. All git transfer must be https. Am I right?

I'm not in China: https://cirosantilli.com/china-dictatorship/has-ciro-santilli-you-ever-lived-in-china

I'm not 100% certain how important going through a third server is, this is beyond my domain of expertise.

You obviously want encryption at all steps, i.e. HTTPS. And HTTPS alone already cannot be broken by anybody, the CCP won't be able to read the message.

It all depends on how likely you think that a timing attack is: https://book.cyberyozh.com/timing-attack-how-special-services-deanonymize-users-messengers/ If you VPN into Tor however, a timing attack feels very unlikely given that each circuit will have a different timing.

And please don't leak your ID through git log like terminus 2049 did: https://cirosantilli.com/china-dictatorship/terminus2049 !! So newb.

I also recommend trying to contact the following people/groups:

- https://cirosantilli.com/china-dictatorship/duty-machine They seem to exclusively use a bot. 
- https://github.com/chinatimeline/chinatimeline.github.io also bot related stuff.

Perhaps just HTTPS and then using a bot through a private github repo is the best option.

Research things thoroughly before doing anything.

If you would like to publish or highlight a limited number of things, feel free to contact me privately on protonmail: https://cirosantilli.com/contact and I might publish it myself.

Stay safe.

kain-hao, 2022-05-13:
1450这活干的是否太轻松了
Taiwanese paid posters even dont fix Chinese grammatical mistakes when they cntl-c/v kek'd

aLoNeIT, 2022-05-31:
沙雕，1450，拿钱办事，背离事实在说话。
漂亮国最新枪击案，20多人付出生命，“真人权”！
后续开发儿童Gun也是脑回路清奇！

cirosantilli, 2022-05-31:
Hi aLoNeIT,

> 沙雕，1450，拿钱办事，背离事实在说话。 漂亮国最新枪击案，20多人付出生命，“真人权”！ 后续开发儿童Gun也是脑回路清奇！

# 新疆维吾尔族人在公共汽车上被迫流离失所 Forced displacement of Uyghurs from Xinjiang across China in busses (2020)

https://www.codastory.com/authoritarian-tech/china-uyghur-migration/

https://cirosantilli.com/china-dictatorship/#forced-displacement-of-uyghurs-across-china-2020

https://user-images.githubusercontent.com/1429315/114305864-d78b3900-9ad1-11eb-9b23-be4db4ae1fa7.mp4

https://user-images.githubusercontent.com/1429315/114305839-c3473c00-9ad1-11eb-994a-e553d80181e9.mp4

https://user-images.githubusercontent.com/1429315/114305828-b6c2e380-9ad1-11eb-96f0-b65910026d24.mp4

https://user-images.githubusercontent.com/1429315/114305847-ce9a6780-9ad1-11eb-80cd-0ed9ffc1cbc6.mp4

# Merdan Ghappar's prison video (2020, 麥爾丹·阿巴的监狱视频)

https://cirosantilli.com/china-dictatorship/#merdan-ghappars-video-2020

https://user-images.githubusercontent.com/1429315/122738464-e92f5080-d279-11eb-8b96-5524d179b11a.mp4

https://user-images.githubusercontent.com/1429315/122738573-0532f200-d27a-11eb-8494-0f9aaf78cf0f.mp4

# 社会主义核心价值观，一定维护民族团结 Uyghur brainwashing: Socialist core values maintain ethnic harmony UNCONFIRMED

https://user-images.githubusercontent.com/1429315/114300978-920f4180-9aba-11eb-903d-23a145b1c5ba.mp4

https://www.youtube.com/watch?v=AxnS4z-cSlI

https://twitter.com/__Inty__/status/1381002964551168000

https://cirosantilli.com/china-dictatorship/#xinjiang-re-education-camps

#  维吾尔人在拘留营唱着快乐的歌 Uyghurs singing a happy song at internment camp 

https://www.youtube.com/watch?t=432&v=WmId2ZP3h0c

https://cirosantilli.com/china-dictatorship/#xinjiang-re-education-camps

https://user-images.githubusercontent.com/1429315/103039458-deb12a00-4568-11eb-9281-7cb8550666ca.mp4

#  2018年维吾尔族人在喀什营地打乒乓球跳快乐舞 Uyghurs playing ping pong and dancing a happy dance in a Kashgar camp in 2018

https://mobile.twitter.com/HuXijin_GT/status/1055013584806256644  by Hu Xijin 胡锡进

https://user-images.githubusercontent.com/1429315/130763853-b4541f8f-1b62-46ba-b617-f4ca60fbeae7.mp4

https://user-images.githubusercontent.com/1429315/130763814-33f8b836-ae8a-4764-892a-496430844378.mp4

# Uyghur children marched with large bags UNCONFIRMED, @MaD DaN, 伯干村 Bogancun, Xinjiang 诗书达理之山人刨作原声

https://user-images.githubusercontent.com/1429315/114303176-b6701b80-9ac4-11eb-8c27-38e2413209ab.mp4

https://twitter.com/Uyghurian/status/1380992770823835664

https://cirosantilli.com/china-dictatorship/#xinjiang-re-education-camps

# Teacher in Xinjiang slaps Uyghur student on the face 11 times 新疆老师打维族学生11个耳光 （建设美丽新疆）(UNCONFIRMED)

https://user-images.githubusercontent.com/1429315/117796953-7331d580-b247-11eb-9af8-6042470db542.mp4

https://www.reddit.com/r/China/comments/n99fe9/teacher_in_xinjiang_school_slaps_a_uyghur_student/

This video cannot be uploaded to YouTube because of their child safety community guidelines. It was automatically flagged, and my appeal was declined by a human. Basically it's bad for advertisers, and some politician might get sad. They prefer dictatorship to truth.

# 新疆维吾尔再教育營 Xinjiang re-education camps

https://cirosantilli.com/china-dictatorship/#xinjiang

https://cirosantilli.com/china-dictatorship/#xinjiang-re-education-camps

Xinjiang Data Project screenshot of camp map 2021 jpg

[image]

Xinjiang camp in camp jpg

[image]

Uyghur playing guitar jpg

[image]

Xinjiang prisoners sitting identified jpeg

[image]

Dabancheng reeducation camp jpg

[image]

Artux city reeducation camp webp

[image]

Uyghur sale website jpg

[image]

Uyghur workers waving China flag jpg

[image]

Xinjiang cotton vs American plantation from 1908 jpg

[image]

Racism in the West vs China George Floyd vs Xinjiang jpg

[image]

Xinjiang I won jpeg

[image]

https://cirosantilli.com/china-dictatorship/#holocaust-denial

Blood Cotton Initiative by Wuheqilin holocaust denial jpg

[image]

# Xinjiang outside reeducation camps

Xinjiang chained butcher knives jpg

[image]