cirosantilli/linux-kernel-module-cheat #173: ln -sf ../configure xx:  no such file or directory

BeenLi, 2022-04-23:
`./run-docker sh -- ./build --download-dependencies qemu-buildroot`
[image]

BeenLi, 2022-04-24:
```
 cd /root/lkmc/submodules/buildroot && \
  FORCE_UNSAFE_CONFIGURE=1 \
  make \
  'LKMC_PARSEC_BENCHMARK_SRCDIR="/root/lkmc/submodules/parsec-benchmark"' \
  O=/root/lkmc/out.docker/buildroot/build/default/x86_64 \
  V=0 \
  all \
;
/usr/bin/make -j1 O=/root/lkmc/out.docker/buildroot/build/default/x86_64 HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
make[1]: Entering directory '/root/lkmc/submodules/buildroot'
  GEN     /root/lkmc/out.docker/buildroot/build/default/x86_64/Makefile
make[1]: Leaving directory '/root/lkmc/submodules/buildroot'
>>> host-gcc-initial custom Configuring
mkdir -p /root/lkmc/out.docker/buildroot/build/default/x86_64/build/host-gcc-initial-custom/build
ln -sf ../configure /root/lkmc/out.docker/buildroot/build/default/x86_64/build/host-gcc-initial-custom/build/configure
```

cirosantilli, 2022-04-24:
Hi there,

I just ran the commands on b0f4f5ec4aac29d96aae2b0d1ad51da26f797218 and they finished fine, I can't reproduce unfortunately:

```
./setup
./run-docker create && \
./run-docker sh -- ./build --download-dependencies qemu-buildroot
./run-docker
```

I suspect that the problem is that something is not being downloaded properly due to the firewall. Can you try nuking everything and doing the build from a VPN?

Buildroot downloads all its sources files from different sources, which leads to a clean build, but also makes things difficult for people behind firewalls.

BeenLi, 2022-04-24:
I've checked all the submodules and they're all downloaded completely, It's been bothering me for two days. Can you give me some advices?

cirosantilli, 2022-04-24:
I don't think its the submodules that are failing, but rather the sources that Buildroot downloads, e.g. gcc, glibc, and many many others. The VPN is really the best I can think of.

It would also be good if you can provide more logs, and as text rather than as an image.

BeenLi, 2022-04-24:
/usr/bin/make -j1 O=/root/lkmc/out.docker/buildroot/build/default/x86_64 HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
make[1]: Entering directory '/root/lkmc/submodules/buildroot'
  GEN     /root/lkmc/out.docker/buildroot/build/default/x86_64/Makefile
make[1]: Leaving directory '/root/lkmc/submodules/buildroot'
>>> host-gcc-initial custom Configuring
mkdir -p /root/lkmc/out.docker/buildroot/build/default/x86_64/build/host-gcc-initial-custom/build
ln -sf ../configure /root/lkmc/out.docker/buildroot/build/default/x86_64/build/host-gcc-initial-custom/build/configure
(cd /root/lkmc/out.docker/buildroot/build/default/x86_64/build/host-gcc-initial-custom/build && rm -rf config.cache; PATH="/root/lkmc/out.docker/buildroot/build/default/x86_64/host/bin:/root/lkmc/out.docker/buildroot/build/default/x86_64/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" PKG_CONFIG="/root/lkmc/out.docker/buildroot/build/default/x86_64/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/root/lkmc/out.docker/buildroot/build/default/x86_64/host/lib/pkgconfig:/root/lkmc/out.docker/buildroot/build/default/x86_64/host/share/pkgconfig" AR="/usr/bin/ar" AS="/usr/bin/as" LD="/usr/bin/ld" NM="/usr/bin/nm" CC="/root/lkmc/out.docker/buildroot/build/default/x86_64/host/bin/ccache /usr/bin/gcc" GCC="/root/lkmc/out.docker/buildroot/build/default/x86_64/host/bin/ccache /usr/bin/gcc" CXX="/root/lkmc/out.docker/buildroot/build/default/x86_64/host/bin/ccache /usr/bin/g++" CPP="/usr/bin/cpp" OBJCOPY="/usr/bin/objcopy" RANLIB="/usr/bin/ranlib" CPPFLAGS="-I/root/lkmc/out.docker/buildroot/build/default/x86_64/host/include" CFLAGS="-O2 -I/root/lkmc/out.docker/buildroot/build/default/x86_64/host/include" CXXFLAGS="-O2 -I/root/lkmc/out.docker/buildroot/build/default/x86_64/host/include" LDFLAGS="-L/root/lkmc/out.docker/buildroot/build/default/x86_64/host/lib -Wl,-rpath,/root/lkmc/out.docker/buildroot/build/default/x86_64/host/lib" INTLTOOL_PERL=/usr/bin/perl CFLAGS="-O2 -I/root/lkmc/out.docker/buildroot/build/default/x86_64/host/include" LDFLAGS="-L/root/lkmc/out.docker/buildroot/build/default/x86_64/host/lib -Wl,-rpath,/root/lkmc/out.docker/buildroot/build/default/x86_64/host/lib" MAKEINFO=missing CFLAGS_FOR_TARGET="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -O0 -g3  -Wno-error" CXXFLAGS_FOR_TARGET="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -O0 -g3 " CONFIG_SITE=/dev/null ./configure --prefix="/root/lkmc/out.docker/buildroot/build/default/x86_64/host" --sysconfdir="/root/lkmc/out.docker/buildroot/build/default/x86_64/host/etc" --localstatedir="/root/lkmc/out.docker/buildroot/build/default/x86_64/host/var" --enable-shared --disable-static --disable-gtk-doc --disable-gtk-doc-html --disable-doc --disable-docs --disable-documentation --disable-debug --with-xmlto=no --with-fop=no --disable-nls   --target=x86_64-buildroot-linux-gnu --with-sysroot=/root/lkmc/out.docker/buildroot/build/default/x86_64/host/x86_64-buildroot-linux-gnu/sysroot --enable-__cxa_atexit --with-gnu-ld --disable-libssp --disable-multilib --disable-decimal-float --with-gmp=/root/lkmc/out.docker/buildroot/build/default/x86_64/host --with-mpc=/root/lkmc/out.docker/buildroot/build/default/x86_64/host --with-mpfr=/root/lkmc/out.docker/buildroot/build/default/x86_64/host --with-pkgversion="Buildroot -gccebc330" --with-bugurl="http://bugs.buildroot.net/" --without-zstd --enable-libquadmath --enable-tls --enable-plugins --enable-lto --enable-threads --with-isl=/root/lkmc/out.docker/buildroot/build/default/x86_64/host --with-arch="nocona" --enable-languages=c --disable-shared --without-headers --disable-threads --with-newlib --disable-largefile  )
/bin/bash: ./configure: No such file or directory
make: *** [package/pkg-generic.mk:240: /root/lkmc/out.docker/buildroot/build/default/x86_64/build/host-gcc-initial-custom/.stamp_configured] Error 127
Traceback (most recent call last):
  File "./build", line 680, in <module>
    Main().cli()
  File "/root/lkmc/cli_function.py", line 267, in cli
    exit_status = self.cli_noexit(*args, **kwargs)
  File "/root/lkmc/cli_function.py", line 258, in cli_noexit
    return self._do_main(vars(args))
  File "/root/lkmc/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/root/lkmc/common.py", line 1542, in main
    ret = self.timed_main()
  File "./build", line 675, in timed_main
    ret = component.build(self.env['arch'])
  File "./build", line 54, in build
    return self.build_callback()
  File "./build", line 473, in f
    return lkmc.import_path.import_path_main(component_file)(**args)
  File "/root/lkmc/common.py", line 739, in __call__
    return super().__call__(**kwargs)
  File "/root/lkmc/cli_function.py", line 149, in __call__
    return self._do_main(kwargs)
  File "/root/lkmc/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/root/lkmc/common.py", line 1542, in main
    ret = self.timed_main()
  File "/root/lkmc/common.py", line 2061, in timed_main
    return self.build()
  File "/root/lkmc/build-buildroot", line 146, in build
    self.sh.run_cmd(
  File "/root/lkmc/shell_helpers.py", line 457, in run_cmd
    raise e
Exception: Command exited with status: 2

BeenLi, 2022-04-25:
There seems to be a buildrood's ` package/glibc/host-gcc-initial` problem

cirosantilli, 2022-04-25:
Yes, it is a bit weird since I do track gcc/glibc in submodules.

`-custom` indicates that it is coming from the submodule.

So the file submodules/gcc/configure exists? But out.docker/buildroot/build/default/x86_64/build/host-gcc-initial-custom/configure doesn't?

Because both exist for me. It should be copying from the submodule to the build.

And you get that problem after a fully clean clone?

BeenLi, 2022-04-25:
Thank you for your time. You are right. It is my VPN problem. I was shocked when I turned on the -V, I find that I couldn't download  some packages the buildroot needs

closed 2022-04-25 by BeenLi