cirosantilli/linux-kernel-module-cheat #174: Plan to support risC-V architecture?

BeenLi, 2022-04-28:


cirosantilli, 2022-04-28:
Send PR and I'll gladly merge. I have booted RISC-V buildroot on QEMU in the past, so definitely doable, wasn't hard. Try this config: https://github.com/buildroot/buildroot/blob/master/configs/qemu_riscv64_virt_defconfig

BeenLi, 2022-04-29:
ok, thanks. I'll try it when I have time.

closed 2022-04-29 by BeenLi

cirosantilli, 2022-04-29:
Some tips if you try:

- first use this minimal procedure but with the riscv64 config: https://stackoverflow.com/questions/47557262/how-to-download-the-torvalds-linux-kernel-master-recompile-it-and-boot-it-wi/49349237#49349237
- do it on the same version of buildroot as in this repo, currently 2020.08.1 + two minor patches, hopefully we can avoid changing buildroot version as it can break other things unexpectedly