cirosantilli/linux-kernel-module-cheat #215: ubuntu 22.04 host  ./build --download-dependencies qemu-buildroot fails on    55 | #elif HAVE_LIBSIGSEGV && SIGSTKSZ < 16384

bob330, 2022-06-05:
The error message:

```
In file included from /usr/include/signal.h:328,
                 from ./signal.h:52,
                 from c-stack.c:49:
c-stack.c:55:26: error: missing binary operator before token "("
   55 | #elif HAVE_LIBSIGSEGV && SIGSTKSZ < 16384
      |                          ^~~~~~~~
make[5]: *** [Makefile:1915: c-stack.o] Error 1
make[5]: *** Waiting for unfinished jobs....
make[4]: *** [Makefile:1674: all] Error 2
make[3]: *** [Makefile:1572: all-recursive] Error 1
make[2]: *** [Makefile:1528: all] Error 2
make[1]: *** [package/pkg-generic.mk:250: /home/charlie/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-m4-12
make: *** [Makefile:84: _all] Error 2
Traceback (most recent call last):
  File "/home/charlie/linux-kernel-module-cheat/./build", line 680, in <module>
    Main().cli()
  File "/home/charlie/linux-kernel-module-cheat/cli_function.py", line 267, in cli
    exit_status = self.cli_noexit(*args, **kwargs)
  File "/home/charlie/linux-kernel-module-cheat/cli_function.py", line 258, in cli_noexit
    return self._do_main(vars(args))
  File "/home/charlie/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/charlie/linux-kernel-module-cheat/common.py", line 1555, in main
    ret = self.timed_main()
  File "/home/charlie/linux-kernel-module-cheat/./build", line 675, in timed_main
    ret = component.build(self.env['arch'])
  File "/home/charlie/linux-kernel-module-cheat/./build", line 54, in build
    return self.build_callback()
  File "/home/charlie/linux-kernel-module-cheat/./build", line 473, in f
    return lkmc.import_path.import_path_main(component_file)(**args)
  File "/home/charlie/linux-kernel-module-cheat/common.py", line 752, in __call__
    return super().__call__(**kwargs)
  File "/home/charlie/linux-kernel-module-cheat/cli_function.py", line 149, in __call__
    return self._do_main(kwargs)
  File "/home/charlie/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/charlie/linux-kernel-module-cheat/common.py", line 1555, in main
    ret = self.timed_main()
  File "/home/charlie/linux-kernel-module-cheat/common.py", line 2074, in timed_main
    return self.build()
  File "/home/charlie/linux-kernel-module-cheat/build-buildroot", line 146, in build
    self.sh.run_cmd(
  File "/home/charlie/linux-kernel-module-cheat/shell_helpers.py", line 457, in run_cmd
    raise e
Exception: Command exited with status: 2
charlie@cod:~/linux-kernel-module-cheat$
```

cirosantilli, 2022-06-05:
Buildroot build breaks on every Ubuntu update as mentioned at https://github.com/cirosantilli/linux-kernel-module-cheat/tree/3accfcda217e4a121d1948ee3a01eee99db5f47a#qemu-buildroot-setup-getting-started

 You have to either use Docker https://github.com/cirosantilli/linux-kernel-module-cheat/tree/3accfcda217e4a121d1948ee3a01eee99db5f47a#docker or patch Buildroot. I accept a buildroot patch. It is almost always a single cherry pick from master.

oaken-source, 2022-06-15:
it seems like this is the relevant buildroot commit:
https://git.buildroot.net/buildroot/commit/?id=5a9504831f3fa1ef3be334036c93da30150fde55

oaken-source, 2022-06-16:
m4 finished building, but there were other issues after that for other buildroot packages. but I managed to complete the build with more patches. I did not build on Ubuntu though, I built on Archlinux.

cirosantilli, 2022-06-16:
Ahh, OK. Thanks for leaving the comment anyways though. Do send the patchlist if you can though anyways, I suspect it might be similar.