cirosantilli/linux-kernel-module-cheat #225: using docker fails with command not found

ysimhony, 2022-10-09:
acovs@yacovs:~/Desktop/linux-kernel-module-cheat$ sudo apt-get install docker 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
docker is already the newest version (1.5-2).
0 upgraded, 0 newly installed, 0 to remove and 141 not upgraded.
yacovs@yacovs:~/Desktop/linux-kernel-module-cheat$ ./setup 
+ '[' 0 -eq 1 ']'
+ y=
+ '[' -f /.dockerenv ']'
+ sudo=sudo
+ sudo apt-get update
Hit:1 http://security.ubuntu.com/ubuntu jammy-security InRelease
Hit:2 http://us.archive.ubuntu.com/ubuntu jammy InRelease
Hit:3 http://us.archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:4 http://us.archive.ubuntu.com/ubuntu jammy-backports InRelease
Reading package lists... Done
+ '[' -f /.dockerenv ']'
+ git=git
+ sudo apt-get install git python3 python3-pip python3-distutils
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-pip is already the newest version (22.0.2+dfsg-1).
git is already the newest version (1:2.34.1-1ubuntu1.4).
python3 is already the newest version (3.10.6-1~22.04).
python3-distutils is already the newest version (3.10.6-1~22.04).
0 upgraded, 0 newly installed, 0 to remove and 141 not upgraded.
+ python3 -m pip install --user -r requirements.txt
Requirement already satisfied: Cython==0.29.15 in /home/yacovs/.local/lib/python3.10/site-packages (from -r requirements.txt (line 1)) (0.29.15)
Requirement already satisfied: china-dictatorship==0.0.74 in /home/yacovs/.local/lib/python3.10/site-packages (from -r requirements.txt (line 2)) (0.0.74)
Requirement already satisfied: pexpect==4.6.0 in /home/yacovs/.local/lib/python3.10/site-packages (from -r requirements.txt (line 3)) (4.6.0)
Requirement already satisfied: ptyprocess>=0.5 in /usr/lib/python3/dist-packages (from pexpect==4.6.0->-r requirements.txt (line 3)) (0.7.0)
yacovs@yacovs:~/Desktop/linux-kernel-module-cheat$ ./run-docker create
+ sudo docker build -t lkmc .
sudo: docker: command not found
Traceback (most recent call last):
  File "/home/yacovs/Desktop/linux-kernel-module-cheat/./run-docker", line 69, in <module>
    cmd_action_map[args.cmd](args.args)
  File "/home/yacovs/Desktop/linux-kernel-module-cheat/./run-docker", line 57, in <lambda>
    'create': lambda args: create(args),
  File "/home/yacovs/Desktop/linux-kernel-module-cheat/./run-docker", line 17, in create
    sh.run_cmd(docker + ['build', '-t', image_name, '.', LF])
  File "/home/yacovs/Desktop/linux-kernel-module-cheat/shell_helpers.py", line 457, in run_cmd
    raise e
Exception: Command exited with status: 1

My ubuntu version is:
yacovs@yacovs:~/Desktop/linux-kernel-module-cheat$ cat /etc/os-release 
PRETTY_NAME="Ubuntu 22.04.1 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.1 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

cirosantilli, 2022-10-09:
Weird. So if you run `sudo docker` or `docker` on the host terminal manually the command exists?

ysimhony, 2022-10-09:
yacovs@yacovs:~/Desktop/linux-kernel-module-cheat$ sudo apt-get install docker
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
docker is already the newest version (1.5-2).
0 upgraded, 0 newly installed, 0 to remove and 94 not upgraded.
yacovs@yacovs:~/Desktop/linux-kernel-module-cheat$ sudo docker 
sudo: docker: command not found

ysimhony, 2022-10-12:
Hi @cirosantilli,

Any suggestion?

Shall I try use ubuntu 20.04 instead?

Thanks,

cirosantilli, 2022-10-12:
Just in case: are you sure you are not running docker inside docker?

If you can't install and run docker on Ubuntu host which is what it seems like, then that is a big problem that I can't understand.

cirosantilli, 2022-10-12:
Is your PATH messed up? Does `/usr/bin/docker` exist?

ysimhony, 2022-10-14:
Hi @cirosantilli 

This is what I found:

yacovs@yacovs:~/linux-5.8.1_2/linux-5.8.1$ sudo find / -name docker
[sudo] password for yacovs: 
find: ‘/run/user/1000/doc’: Permission denied
find: ‘/run/user/1000/gvfs’: Permission denied
/usr/share/doc/docker

coltensu, 2022-12-28:
I have the same problem，using "sudo snap install docker" can solve this issue.

zhh123zhh, 2023-03-24:
你妈死了嗷

zhh123zhh, 2023-03-24:
傻逼