cirosantilli/linux-kernel-module-cheat #262: add USER for docker

atharvanan1, 2023-08-08:
I noticed that using `docker` to do first build, will result in cloning the submodules with `root` owner. Do we want to add `USER` setting in the Dockerfile to be able to clone those with `USER` owner? This proves editing the submodules a bit difficult.

Not a high priority.

cirosantilli, 2023-08-09:
Hi. Yes, I would like to do that, not only for submodules, but also for the entire out/ directory. This is mentioned at: https://cirosantilli.com/linux-kernel-module-cheat/docker "TODO make files created inside Docker be owned by the current user in host instead of root". If you find a way, do let me know.

atharvanan1, 2023-08-22:
Hi! Thanks for your input. I just read this, I'll take a look.