cirosantilli/linux-kernel-module-cheat #312: Issue with simSeconds variable

fluctlights, 2024-07-01:
Hello:

I am new on gem5 and I wanted to properly understand what the simSeconds variable represents. I have been using it to execute several synthetic benchmarks (Dhrystone, Whetstone) on a Deriv03CPU model, and everything went fine.

The thing is that while looking at stats.txt file I get a real low simSeconds value
As I understand, simSeconds represents the seconds simulated, so it represents the benchmark time. 
Please correct me if I am wrong.

To get the stats I use the m5_addr version of resetstats at the start of it and m5_addr version of dumpstats at the end

When I execute this on a real hardware platform (Raspberry Pi 4B) I get a higher execution time, so I dont really understand why simSeconds is so low.

Thank you in advance.

cirosantilli, 2024-07-02:
Hi.

I haven't touched gem5 in a while and don't have much time to look deeply, but from a quick grep, yes it is the real time simulated, and furthermore it is derived from `simTicks`:

```
src/sim/root.cc:57:    simSeconds(this, "sim_seconds", "Number of seconds simulated"),
src/sim/root.cc:90:    simSeconds = simTicks / simFreq;
```

which is the number of CPU ticks, which is what most people usually use, since the simFreq is somewhat arbitrary.

If the value is different compared to real harware, that is not surprising unfortunately. The built-in gem5 models are very generic/simplistic, and don't precisely model any specific real hardware in particular.

For closed source hardware like x86/arm, vendors won't ever publish an accurate CPU model for example to protect their IP, so the only alternative is for people to reverse engineer stuff. For open RISC-V CPUs they could, but I'm not sure how far they've gone.

In any case, even with open hardware, modelling is a hard task that requires lots of attention and trial and error. A lot can go wrong. Even if the CPU model is accurate ,maybe there is something else very important that is wrong, like memory timings etc. Which gem5 attempts to model, but yet another hw you need to carefully model

fluctlights, 2024-07-03:
Hi:

Thank you for answering! So if I understood correctly, simTicks is the real base of everything. That makes real sense since while looking at the stats the simSeconds was the division by 1e12 of the simulated ticks.

Let me explain a bit further my problem: I want to compare both simulated and real hardware platforms, so I need to get the execution times. This is because I want to model the energy that will consume a set of real devices that are similar in hardware architectures (RPi3, RPi4) via performing gem5 hardware simulations. I have a dynamic and static power models so, given the execution time, will retrieve the energy consumption. As the simulated results are below the real execution times, it might be, as you said, not properly simulated, like memory timings (which they are in the simulation as generic Timing mode). I abstracted a lot of things since I am not a hardware guru: that also means reverse engineering is not viable haha.

I guess I will keep the times to make the comparison and document why is not with the expected accuracy. This is for a master's dissertation, so maybe this approach is fine at this point?

cirosantilli, 2024-07-03:
OK I see. It's really hard to answer if gem5 will be useful or not. You basically just have to run tests and see if things correlate at all with real hardware. And if they don't , it can be very hard to understand why.

I assume you are using some of the following power funcationalities in gem5: https://www.gem5.org/documentation/learning_gem5/part2/arm_power_modelling/ | https://www.gem5.org/documentation/general_docs/thermal_model as well right? I also just Googled into: https://eprints.soton.ac.uk/412825/1/patmos_Camera_Ready.pdf

I'd also consider pinging the Arm people that work on gem5 via git log since given that Arm is particularly interested in power consumption, they might have more info. I worked with some of these dudes in the past... just I never did power research so I don't know specifics.

Overall its a bit sad to waste valuable research time because of closed source stuff. Maybe there isn't a solution however in the current situation.

cirosantilli, 2024-07-03:
Also git blame any power modelling source code to try and find people.